jameszah/ESP32-CAM-Video-Telegram #5: Stack smashing protect failure!

tezmo, 2021-08-06:
Hej!
Great project! I'm loving the addition of video, (but right now do not have a PIR attached, not sure if that is needed)?

My issue is: Stack smashing protect failure!

```
---------------------------------
ESP32-CAM Video-Telegram pir-cam 8.8
---------------------------------
Connecting Wifi: MY SSID
.......mDNS responder started 'deskpir'

WiFi connected
IP address: 192.168.68.148

Stack smashing protect failure!

abort() was called at PC 0x40165f37 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f950:0x3ffb1e70 0x4008fbc9:0x3ffb1e90 0x40165f37:0x3ffb1eb0 0x400d2d41:0x3ffb1ed0 0x400d2e4b:0x3ffb1f30 0x400daab2:0x3ffb1fb0 0x4009185a:0x3ffb1fd0

Rebooting...
```

Any idea how to solve? Didn't adjust any of the code at all. 
Using an AI thinker ESP32 board, runs the normal Telegram Photo bot fine.

jameszah, 2021-08-06:
hmmm ... after the wifi setup ...

```
String stat = "Reboot\nDevice: " + devstr + "\nVer: " + String(vernum) + "\nRssi: " + String(WiFi.RSSI()) + "\nip: " +  WiFi.localIP().toString() + "\n/start";
bot.sendMessage(chat_id, stat, "");
```

Might be a string issue in the that "stat" line - are all those variables set?

Then the bot.sendMessage which launches wificlientsecrure (which needs bags of memory).  

You could throw in this line in a few places to see where you are running out of memory.

```
Serial.printf("Internal Total heap %d, internal Free Heap %d\n", ESP.getHeapSize(), ESP.getFreeHeap());
```

Are you using Arduino 1.8.13 and esp32-arduino 1.06???

That "Stack smashing " message is not familiar to me -- running out of heap is familiar, but not that message.

tezmo, 2021-08-06:
Thanks for the quick response!
`Might be a string issue in the that "stat" line - are all those variables set?`
Yes, all are set fine, also ruled out hardware failure now, by testing it on a 2nd board (same results).

`Are you using Arduino 1.8.13 and esp32-arduino 1.06???`
Yes to both

And i put in a bunch of those.. and i don't get it.. :/
```
Serial.printf("1 Internal Total heap %d, internal Free Heap %d\n", ESP.getHeapSize(), ESP.getFreeHeap());
  avi_buf_size = 3000 * 1024; // = 3000 kb = 60 * 50 * 1024;
  idx_buf_size = 200 * 10 + 20;
  psram_avi_buf = (uint8_t*)ps_malloc(avi_buf_size);
  if (psram_avi_buf == 0) Serial.printf("psram_avi allocation failed\n");
  psram_idx_buf = (uint8_t*)ps_malloc(idx_buf_size); // save file in psram
  Serial.printf("2 Internal Total heap %d, internal Free Heap %d\n", ESP.getHeapSize(), ESP.getFreeHeap());
  if (psram_idx_buf == 0) Serial.printf("psram_idx allocation failed\n");
  if (!setupCamera()) {
    Serial.println("Camera Setup Failed!");
    while (true) {
      delay(100);
      Serial.printf("4 Internal Total heap %d, internal Free Heap %d\n", ESP.getHeapSize(), ESP.getFreeHeap());
    }
    Serial.printf("5 Internal Total heap %d, internal Free Heap %d\n", ESP.getHeapSize(), ESP.getFreeHeap());
  }
  Serial.printf("6 Internal Total heap %d, internal Free Heap %d\n", ESP.getHeapSize(), ESP.getFreeHeap());
  bool wifi_status = init_wifi();
  Serial.printf("7 Internal Total heap %d, internal Free Heap %d\n", ESP.getHeapSize(), ESP.getFreeHeap());
  // Make the bot wait for a new message for up to 60seconds
  //bot.longPoll = 60;
  bot.longPoll = 5;

  client.setInsecure();
  Serial.printf("8 Internal Total heap %d, internal Free Heap %d\n", ESP.getHeapSize(), ESP.getFreeHeap());
  setupinterrupts();
```

with the output (the 3 one is on the last line if "init_wifi" after the serial print of the localIP).

```
---------------------------------
ESP32-CAM Video-Telegram pir-cam 8.8
---------------------------------
1 Internal Total heap 301240, internal Free Heap 273164
2 Internal Total heap 301240, internal Free Heap 273164
6 Internal Total heap 300728, internal Free Heap 245892
Connecting Wifi: JustGuests
......mDNS responder started 'Cammie'

WiFi connected
IP address: 192.168.68.149
3 Internal Total heap 298848, internal Free Heap 187760

Stack smashing protect failure!

abort() was called at PC 0x4016613b on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f950:0x3ffb1e70 0x4008fbc9:0x3ffb1e90 0x4016613b:0x3ffb1eb0 0x400d2d75:0x3ffb1ed0 0x400d2f0f:0x3ffb1f30 0x400dac2e:0x3ffb1fb0 0x4009185a:0x3ffb1fd0

Rebooting...
```

jameszah, 2021-08-07:
https://github.com/jameszah/ESP32-CAM-Video-Telegram/blob/74088e263956f845a32ab7be890538385d746c1e/ESP32-CAM-Video-Telegram_8.8x.ino#L993

What is your timezone?  Is it longer than 40 chars?

I use this:

String TIMEZONE = "MST7MDT,M3.2.0/2:00:00,M11.1.0/2:00:00" ; //"GMT0BST,M3.5.0/01,M10.5.0/02";

Maybe my GMT timezone example string is wrong.  Try something simple like "GMT".
Might be something strange returning from the configtime call - which would be asynchronous to the flow of code ???

My timezone is 38 long, and char array is defined as 40 ???

It prints the ip,  but doesn't make it back from the init_wifi()?

jameszah, 2021-08-07:
Or take all the time stuff out as an experiment.

tezmo, 2021-08-07:
You found it! I increased the tzchar size to 50 and it started working :) My timezone is in the EU en it's MUCH longer.
Thanks for the support! 

Great stuff, thanks so much for building this!

closed 2021-08-07 by tezmo