jameszah/ESP32-CAM-Video-Telegram #8: non-void function error

MikeGitHubRPi, 2021-11-16:
Hello James, thank you very much for your code.

I have a problem when compiling in Arduino IDE:

In function 'esp_err_t start_avi()':
warning: no return statement in function returning non-void [-Wreturn-type]
 } // end of start avi
 ^

In function 'esp_err_t another_save_avi(camera_fb_t*)':
warning: no return statement in function returning non-void [-Wreturn-type]
 } // end of another_pic_avi
 ^

In function 'esp_err_t end_avi()':
warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^

In function 'bool init_wifi()':
warning: control reaches end of non-void function [-Wreturn-type]
   Serial.println(WiFi.localIP());

I've tried rolling back to library versions described in the header but same response.

Any idea about that warning?

Thank you

jameszah, 2021-11-16:
I think you have too many compiler warnings turned on
Try this in Arduino File Preferences ...

[image]

MikeGitHubRPi, 2021-11-16:
Hello James, thank you for your answer.
The problem are not the warnings themselves, I can't start the communication with esp32-cam. I only see this in the serial monitor.

21:37:22.005 -> Connecting Wifi: WIRELESS2
21:37:22.098 -> .......mDNS responder started 'CAM01'
21:37:25.640 -> 
21:37:25.640 -> WiFi connected
21:37:25.640 -> IP address: 192.168.1.110
21:37:25.640 -> mDNS responder started 'CAM01'
21:37:25.640 -> 
21:37:25.640 -> WiFi connected
21:37:25.640 -> IP address: 192.168.1.110
21:37:25.640 -> mDNS responder started 'CAM01'
21:37:25.640 -> 
21:37:25.640 -> WiFi connected
21:37:25.640 -> IP address: 192.168.1.110
21:37:25.640 -> mDNS responder started 'CAM01'

The last three lines repeat continuously.

Thank you

jameszah, 2021-11-16:
Is this the code you have?
You seem to have a loop from 1009 back to 999, without going up to 971 (so not restarting procedure).

What is your timezone?
`String TIMEZONE = "GMT0BST,M3.5.0/01,M10.5.0/02";`

[image]

jameszah, 2021-11-26:
Do you have a space at the beginning of that timezone string?
I'll try to have a look at it today.

MikeGitHubRPi, 2021-11-27:
Hello,

That was a mistake with copy and paste from the code. There is no space.

Thank you

mrWheel, 2022-01-18:
I have the same problem ....

If I comment the mDSN statement's my ESP32 keeps crashing ..

```
---------------------------------
ESP32-CAM Video-Telegram pir-cam 8.8
---------------------------------
Internal Total heap 292384, internal Free Heap 265792
SPIRam Total heap   4192115, SPIRam Free Heap   1118095
Internal Total heap 292120, internal Free Heap 228568
SPIRam Total heap   4192067, SPIRam Free Heap   872287
Camera setup OK!
Connecting Wifi: <myWiFi>
..... Connected!
set TimeZonde .. OK!

WiFi connected
IP address: 192.168.x.y
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d4464  PS      : 0x00060430  A0      : 0x800d453c  A1      : 0x3ffb2770  
A2      : 0x3ffc87a4  A3      : 0x3ffbacc0  A4      : 0x3ffc867c  A5      : 0x3ffbdc77  
A6      : 0x43ffc000  A7      : 0x3ffc8630  A8      : 0x800d4464  A9      : 0x3ffb2750  
A10     : 0x0000000f  A11     : 0x0000000d  A12     : 0x00000001  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x3ffc8b04  SAR     : 0x0000000c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000001b  LBEG    : 0x4008c8e5  LEND    : 0x4008c8f5  LCOUNT  : 0xffffffff  

Backtrace:0x400d4461:0x3ffb27700x400d4539:0x3ffb27f0 0x400dc6ba:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1420
ho 0 tail 12 room 4
load:0x40078000,len:13540
load:0x40080400,len:3604
entry 0x400805f0
---------------------------------
ESP32-CAM Video-Telegram pir-cam 8.8
---------------------------------
Internal Total heap 292384, internal Free Heap 265792
SPIRam Total heap   4192115, SPIRam Free Heap   1118095
Internal Total heap 292120, internal Free Heap 228568
SPIRam Total heap   4192067, SPIRam Free Heap   872287
Camera setup OK!
Connecting Wifi: <myWiFi>
..... Connected!
set TimeZonde .. OK!

WiFi connected
IP address: 192.168.x.y
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d4464  PS      : 0x00060430  A0      : 0x800d453c  A1      : 0x3ffb2770  
A2      : 0x3ffc87a4  A3      : 0x3ffbacc0  A4      : 0x3ffc867c  A5      : 0x3ffbdc77  
A6      : 0x43ffc000  A7      : 0x3ffc8630  A8      : 0x800d4464  A9      : 0x3ffb2750  
A10     : 0x0000000f  A11     : 0x0000000d  A12     : 0x00000001  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x3ffc8b04  SAR     : 0x0000000c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000001b  LBEG    : 0x4008c8e5  LEND    : 0x4008c8f5  LCOUNT  : 0xffffffff  

Backtrace:0x400d4461:0x3ffb27700x400d4539:0x3ffb27f0 0x400dc6ba:0x3ffb2820 

...
...
```

jameszah, 2022-01-18:
what is your "devname"?  Is it a simple short text string - no spaces, punctuation, etc?
Is the IP address: 192.168.x.y changed ... it has real numbers on the serial monitor?
What is your esp32-arduino version ... like 1.6 or 2.0.2 ?
It crashed when dns is commented out, or commented in ... included?

mrWheel, 2022-01-18:
@jameszah 
devname = “CAM182”
Yes, IP address is a normal all numeric IP address
Esp-core 2.0.1 because 2.0.2 has problems with UniversalTelegramBot

It loops when dns is active like MikeGitHubRPi listed in his second comment.
It crashen when commented out.

Tried various tzChar strings (short (GMT) long, with first char as space and without first char space)

jameszah, 2022-01-18:
try new version 8.9

That non-void function return warning seems to be a new feature of the compiler 1.8.19 or so.
It gives a warning during the compile, but at run time, it keeps re-running the function until it gets a "return true" ???

Added a work-around to the main program for handshake timeout problem to avoid having to go back to 2.0.1

Also looked at the 4 old picture problem in another comment.

I suppose this is way people ask "what compiler, library, hardware, ...."

mrWheel, 2022-01-19:
@jameszah 
Thanks for this great software!

about version 8.9:

```
---------------------------------
ESP32-CAM Video-Telegram pir-cam 8.9
---------------------------------
Pic, len=15581, new fb 3FAFF098
Pic, len=15637, new fb 3FB0E0A8
Pic, len=26751, new fb 3FB1D0B8
Pic, len=26671, new fb 3FB2C0C8
Pic, len=29513, new fb 3FAFF098
Pic, len=29566, new fb 3FB0E0A8
Pic, len=29655, new fb 3FB1D0B8
Connecting Wifi: <MY_SID>
.....mDNS responder started 'ESP32CAM'

WiFi connected
IP address: 192.168.x.y
Setup PIRpin = 1, 1, 1, 1, 1,  

Got a message /start

Got a message /photo

>>>>> Sending, bytes=  28361
********done!

Got a message /photo

>>>>> Sending, bytes=  28117
********
```

in `setup()` I altered this:
```
void setup() {
  Serial.begin(115200);
  Serial.println("---------------------------------");
  Serial.printf("ESP32-CAM Video-Telegram %s\n", vernum);
  Serial.println("---------------------------------");

  pinMode(FLASH_LED_PIN, OUTPUT);
  digitalWrite(FLASH_LED_PIN, flashState); //defaults to low

  pinMode(12, INPUT_PULLUP);        // pull this down to stop recording

  pinMode(33, OUTPUT);             // little red led on back of chip
  digitalWrite(33, LOW);           // turn on the red LED on the back of chip
```
What is `gpio12`? PIRpin?
`gpio33` is the PIR sensor on my device!
better to use #define's for this is different on every board!!

I have added a tab `setting_pins.h` by '[LewisHe](https://github.com/lewisxhe/esp32-camera-series)' (but renamed it `camera_pins.h`) and moved your AI_THINKER defines to the `camera_pins.h` file.
The beginning of the sketch now looks like this:
```
#include <WiFi.h>
#include <WiFiClientSecure.h>
#include "esp_camera.h"

/***************************************
 *  Board select
 **************************************/
/* Select your board here */
// #define T_Camera_JORNAL_VERSION
// #define T_Camera_MINI_VERSION
// #define T_Camera_PLUS_VERSION
// #define T_Camera_V05_VERSION
// #define T_Camera_V16_VERSION
// #define T_Camera_V162_VERSION
 #define T_Camera_V17_VERSION
// #define ESPRESSIF_ESP_EYE
// #define CAMERA_MODEL_AI_THINKER

#include "camera_pins.h"
```
I added #defines for `PIR_PIN` and `FLASH_LED_PIN` for every board type.

also in `setupCamera()`
I added:
```
  s->set_framesize(s, (framesize_t)framesize);
  s->set_quality(s, quality);
  //s->set_vflip(s, 1); //-- flip it <<-- uncomment this if camera is upside-down
```

I have only one problem left: After reboot (or start) the PIR sensor seems alway's be enabled. I have already set it to `false` in every place I can imagine but at no evail. After startup it keeps sending me PIR interrupt messages and clips...
What am I doing wrong?

jameszah, 2022-01-24:
I only  use the esp32-cam module.  I don't have those others to test.

You have to find our pir pin, determine if it active-high or active low, and then modify setupinterupts() for your board (at the link below after pirpin definition)

It prints out the status of your pir at the reboot to show you what your pir is telling you during boot.

Pin 12 is an sd-card pin on esp32-cam -- it does not do anything in this program.

https://github.com/jameszah/ESP32-CAM-Video-Telegram/blob/87c9ffa764ef58648e143ca038c49ba51b8964bf/ESP32-CAM-Video-Telegram_8.9x.ino#L951