jameszah/ESP32-CAM-Video-Telegram #14: Not worked :(

BytEvil, 2022-01-10:
Hi! Thanks for the code. But not worked for me...
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
---------------------------------
ESP32-CAM Video-Telegram pir-cam 8.8
                                    ---------------------------------
psram_avi allocation failed
                           psram_idx allocation failed
                                                      Connecting Wifi: NTGR_F7CF
.....mDNS responder started 'deskpir'

WiFi connected
IP address: 172.16.0.63
Setup PIRpin = 0, 0, 0, 0, 0,
PIR Interupt>> 3
PIR Interupt ... start recording ...
                                     Start the avi ... at 29949
                                                               Framesize 8, quality 10, length 0 seconds

                                                                                                        Starting an avi
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4000c2e4  PS      : 0x00060c30  A0      : 0x800d2638  A1      : 0x3ffdb900
A2      : 0x00000000  A3      : 0x3ffbdbb8  A4      : 0x000000f0  A5      : 0x00000000
A6      : 0x46464952  A7      : 0x000e01d8  A8      : 0x00000000  A9      : 0xffffffdb
A10     : 0x00000002  A11     : 0x3ffc29b5  A12     : 0x00000000  A13     : 0x3ffd8d3c
A14     : 0x3ffc29ac  A15     : 0x00000208  SAR     : 0x0000001e  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x0000000e

ELF file SHA256: 0000000000000000

Backtrace: 0x4000c2e4:0x3ffdb900 0x400d2635:0x3ffdb910 0x4008bdde:0x3ffdb960

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
---------------------------------
ESP32-CAM Video-Telegram pir-cam 8.8
                                    ---------------------------------
psram_avi allocation failed
psram_idx allocation failed
Connecting Wifi: NTGR_F7CF
.................................................................................
```

BytEvil, 2022-01-10:
Problem solved by changing device to AI Thinker ESP32-CAM. Thanks for your good work! ;)

closed 2022-01-10 by BytEvil

magdalinos, 2022-08-31:
I am running to the same issue. Can you please explain where was the change made?
Was it in the Arduino IDE?
In Arduino I am using :
Board : AI Thinker ESP32 CAM
CPU Freq : 240 MHz
Flash Req : 80 MHz
Partition Scheme : HUGE APP 3MB No OTA /  1MB SPIFFS

jameszah, 2022-09-02:
[image]

Are you getting these errors?

What arduino-esp32 version are you using.  I've seen some comments about 2.04 not working correctly.  

This was "Compiled with Arduino 1.8.13 and arduino-esp32 1.0.6 (latest release version) on Jun 10, 2021"

You could try that.  I'll take a look at the 2.00+ version problems.

magdalinos, 2022-09-03:
> Thank you for your reply..
Issue was with available memory as mentioned in the discussions as well.
Looks like there are ESP32 CAMs with 2GB only (?!)