jameszah/ESP32-CAM-Video-Telegram #16: PIR Sensor always active.

cruzer619, 2022-02-03:
Not sure what you want from me but looking at serial monitor output.  
=-=-=-=-=-=-=-=-=-=-=
WiFi connected
IP address: 192.168.x.x
Setup PIRpin = 1, 1, 1, 1, 1,  
PIR Interupt>> 3
PIR Interupt ... start recording ... 
Start the avi ... at 20324
Framesize 8, quality 10, length 0 seconds

Starting an avi 

>>>>> Sending as 512 byte blocks, with jzdelay of 0, bytes=  20232
PIR Interupt>> 3
End the Avi
End of avi - closing the files

*** Video recorded and saved ***

Recorded   148 frames in     5 seconds
File size is 2955432 bytes
Adjusted FPS is 25.15
Max data rate is 498927 bytes/s
Frame duration is 39764 us
Average frame length is 19955 bytes
Writng the index, 148 frames
---
End the avi at 26264.  It was 148 frames, 5940 ms at 29.60 fps...
Deleting the camera task
**********

Sending clip with caption

>>>>> Sending as 512 byte blocks, with a caption, and with jzdelay of 0, bytes=  2957452
*********PIR Interupt>> 3
************PIR Interupt>> 3
********PIR Interupt>> 3
***********PIR Interupt>> 3
*********PIR Interupt>> 3
**********PIR Interupt>> 3
**********PIR Interupt>> 3
***********PIR Interupt>> 3
*********PIR Interupt>> 3
**********PIR Interupt>> 3
*********PIR Interupt>> 3
*********PIR Interupt>> 3
***********PIR Interupt>> 3
************PIR Interupt>> 3
********PIR Interupt>> 3
**********PIR Interupt>> 3
*********PIR Interupt>> 3
******************PIR Interupt>> 3
****done!
PIR Interupt>> 3
PIR Interupt ... start recording ... 
Start the avi ... at 151891
Framesize 8, quality 10, length 0 seconds

Starting an avi 

>>>>> Sending as 512 byte blocks, with jzdelay of 0, bytes=  20571
End the Avi
End of avi - closing the files

*** Video recorded and saved ***

Recorded   145 frames in     5 seconds
File size is 2950512 bytes
Adjusted FPS is 25.17
Max data rate is 508408 bytes/s
Frame duration is 39731 us
Average frame length is 20334 bytes
Writng the index, 145 frames
---
End the avi at 157687.  It was 145 frames, 5796 ms at 29.00 fps...
Deleting the camera task
PIR Interupt>> 3

=-=-=-=-=-=-=-=-=-=-=

I know the PIR sensor is working fine.  I connected it to an Arduino uno using a simple example code.  And works as expected.  Show "High" when active and "LOW" when no movement.

Please help.

Thanks

jameszah, 2022-02-04:
You could put an led on gpio 13 to see if it turns off and on, or just stays on.   

Or print out the status every loop().
`Serial.print( digitalRead(PIRpin) );`

It was on during the boot. --- Setup PIRpin = 1, 1, 1, 1, 1,

This is the only code for the pir -- pin 13, pulled-down, and interrupt on rising edge.

https://github.com/jameszah/ESP32-CAM-Video-Telegram/blob/aa2b8535ef066b2b03a172da3c777a2acd2edc4e/ESP32-CAM-Video-Telegram_8.9x.ino#L951

```
int PIRpin = 13;
pinMode(PIRpin, INPUT_PULLDOWN)
gpio_set_intr_type((gpio_num_t)PIRpin, GPIO_INTR_POSEDGE); 
```

cruzer619, 2022-02-04:
ok will try that when I get home.  Didn't think the problem might be the esp32cam module itself.  As I said before on an Arduino uno, PIR sensor functions correctly.  And as you were saying.. if I put an LED on pin 13 and it stays on.. then esp32cam module is faulty, correct?

Thanks

Rich

cruzer619, 2022-02-09:
ok its definitely not the ESP32.  I went ahead and used a simple code to verify.  And is working as expected.  Maybe its the sensor i'm using thats not working well with your code..  I'm using an HC-SR501.  The bigger ones.  Not the the mini sensors.  Maybe thats the issue???  Thanks again for your time with this.

`const int PIN_TO_SENSOR = 13;
int pinStateCurrent   = LOW;  
int pinStatePrevious  = LOW;  

void setup() {
  Serial.begin(115200);            
  pinMode(PIN_TO_SENSOR, INPUT); 
}

void loop() {
  pinStatePrevious = pinStateCurrent; 
  pinStateCurrent = digitalRead(PIN_TO_SENSOR); 
  Serial.println(pinStateCurrent);

  if (pinStatePrevious == LOW && pinStateCurrent == HIGH) { 
    Serial.println("Motion detected!");
  }
  else
  if (pinStatePrevious == HIGH && pinStateCurrent == LOW) {
    Serial.println("Motion stopped!");
  }
}`

cruzer619, 2022-02-09:
Ok figured out my issue..  was running everything on 3.3v.. as i'm using this on some 18650 batteries..  Once I switched everything to 5v ( rewire PIR direct to 5v, was using VCC to power PIR).  All is working as expected.  Hoping wiring PIR direct to batteries will be fine but not sure if it will work right when it goes below 4v when the batteries start to drain.  Oh well my issue.  Thanks again for your time and effort.

closed 2022-02-09 by cruzer619