jameszah/ESP32-CAM-Video-Telegram #20: Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1).

sssseeeaqqq, 2022-04-17:
21:33:35.153 -> PIR Interupt ... start recording ... Deleting the camera task
21:33:35.624 -> 
21:33:35.624 -> >>>>> Sending as 512 byte blocks, with jzdelay of 0, bytes=  20138
21:33:37.480 -> **********PIR Interupt>> 3
21:34:03.464 -> PIR Interupt ... start recording ... Deleting the camera task
21:34:03.891 -> 
21:34:03.891 -> >>>>> Sending as 512 byte blocks, with jzdelay of 0, bytes=  20111
21:34:05.735 -> **********Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 
21:34:32.627 -> 
21:34:32.627 -> Core  1 register dump:
21:34:32.627 -> PC      : 0x40092066  PS      : 0x00060335  A0      : 0x8009100e  A1      : 0x3ffbf080  
21:34:32.627 -> A2      : 0x3ffb8a00  A3      : 0x3ffb8890  A4      : 0x00000004  A5      : 0xb33fffff  
21:34:32.627 -> A6      : 0x00000001  A7      : 0x00000001  A8      : 0x3ffb8890  A9      : 0x00000018  
21:34:32.627 -> A10     : 0x3ffb8890  A11     : 0x00000018  A12     : 0x3ffc9b44  A13     : 0xb33fffff  
21:34:32.674 -> A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x0000000d  EXCCAUSE: 0x00000006  
21:34:32.674 -> EXCVADDR: 0x00000000  LBEG    : 0x4008ca85  LEND    : 0x4008ca95  LCOUNT  : 0xfffffffb  
21:34:32.674 -> Core  1 was running in ISR context:
21:34:32.674 -> EPC1    : 0x400ed27f  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000
21:34:32.674 -> 
21:34:32.674 -> 
21:34:32.674 -> Backtrace:0x40092063:0x3ffbf0800x4009100b:0x3ffbf0a0 0x4009014b:0x3ffbf0c0 0x400dc9cf:0x3ffbf100 0x400db6a9:0x3ffbf120 0x400db99f:0x3ffbf140 0x400dba65:0x3ffbf160 0x40081639:0x3ffbf180 0x400816e1:0x3ffbf1b0 0x40081708:0x3ffbf1d0 0x40085ead:0x3ffbf200 0x4000bfed:0x3ffb2750 0x40092325:0x3ffb2760 0x400df7ca:0x3ffb2780 0x400dc81f:0x3ffb27b0 0x400db62d:0x3ffb27e0 0x400db6c9:0x3ffb2800 0x400dc502:0x3ffb2820 
21:34:32.722 -> 
21:34:32.722 -> 
21:34:32.722 -> Core  0 register dump:
21:34:32.722 -> PC      : 0x400921ee  PS      : 0x00060035  A0      : 0x8009094d  A1      : 0x3ffbeb70  
21:34:32.722 -> A2      : 0x3ffbf218  A3      : 0x3ffbeb8c  A4      : 0x00060023  A5      : 0xb33fffff  
21:34:32.722 -> A6      : 0x0000cdcd  A7      : 0x0000abab  A8      : 0x0000abab  A9      : 0x0000abab  
21:34:32.769 -> A10     : 0x3ffc8bd4  A11     : 0x00000000  A12     : 0x3ffc8bd0  A13     : 0x00000007  
21:34:32.769 -> A14     : 0x00000005  A15     : 0x00000001  SAR     : 0x00000014  EXCCAUSE: 0x00000006  
21:34:32.769 -> EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

It always reboot when PIR is enabled  and sending a photo ,using Arduino 1.8.19 / esp32-arduino 2.0.2

sssseeeaqqq, 2022-04-17:
I'm using AI-thinker esp32-cam

jameszah, 2022-04-18:
Does it work without the pir?  
Can you send commend to take a picture and receive it?  And a video?
In the example does the first picture (20138 bytes) arrive, and the second fail (20111) ?

sssseeeaqqq, 2022-04-18:
work normally without PIR ,but I need it to do detection.
those picture are arrive but each reboot will reset the setting to default.

sssseeeaqqq, 2022-04-19:
> Does it work without the pir? Can you send commend to take a picture and receive it? And a video? In the example does the first picture (20138 bytes) arrive, and the second fail (20111) ?

work normally without PIR ,but I need it to do detection.
those picture are arrive but each reboot will reset the setting to default.

FBMinis, 2022-10-01:
Which PIR module are you using?

mlegnari, 2022-11-16:
Pin 13 runs low with an internal pulldown resistor sinking it. I recommend the following: solder a 1k resistor on the wire that comes out of the PIR and enters the module, this will reduce the input pin 13 current. The same pin solder a ceramic capacitor from 47 to 100 nf to help reduce noise, for me it worked!