jameszah/ESP32-CAM-Video-Telegram #23: add wifi manager

zilhos, 2022-08-16:
Hello sir, for my work I travel a lot, this project fascinates me. It would be too much to ask that you add wifimanager to be able to modify network and token. I admire your work and thank you very much

jameszah, 2022-08-22:
good idea - ill have a look

zilhos, 2022-08-22:
Thank you for your prompt response, I think you do a very good job. Thank
you

jameszah, 2022-10-17:
check the new version!

FBMinis, 2022-10-17:
Testing, thank you so much!

Useful to open Telegram and look for the bot's ID and Token **before** connecting to the deskpir AP, in order to have access to these keys while already in the wifimanager homepage.

Regarding timed photos, the following worked for me: power on; define photo every 3 minutes by entering /3; entering /entim. A photo is taken 3 minutes after this input, then after a 3min interval, etc.
To change the interval, I just entered /distim, entered the new value like /10; entered /entim.

I will try leaving it running for some days to keep testing, thank you!

jameszah, 2022-10-18:
I think it just keeps track of the last photo, and adds interval to do another one ... so you will often get a photo immediately if it has been shut off.
I had thought about switching it to hd 1280x780 - better pictures but shorter video.

zilhos, 2022-10-18:
excellent work sir jameszah! 
Thanks again for sharing. It is really useful to be able to change the network and the token in this type of projects

FBMinis, 2022-10-19:
I left it working for 48hours and overnight my router rebooted for some unrelated reason (maybe maintenance by my ISP).

Looking at the code, once the bot looked for a new message and realized there was no connection, it called `WiFi.reconnect();` but since the AP wasn't up yet, it called `init_wifi();` and Wifimanager run again trying to connect to saved SSID and Pass. However, 60 seconds passed and my AP wasn't up yet (the router takes a few minutes to be back online), so it went back to the main Config Portal.

```
13:01:26.410 -> >>>>> Sending timed photo, bytes=  21649
13:01:33.380 -> done!

13:04:14.907 -> *wm:[2] [EVENT] WIFI_REASON:  200
13:04:25.485 -> *wm:[2] [EVENT] WIFI_REASON:  201
13:04:25.485 -> *wm:[2] [EVENT] WIFI_REASON: NO_AP_FOUND 
13:04:26.386 -> ***** WiFi rerestart *****
13:04:26.386 -> Good settings in the EPROM 
13:04:26.386 -> *wm:[2] Added Parameter: DevName
13:04:26.386 -> *wm:[2] Added Parameter: chat_id
13:04:26.386 -> *wm:[2] Added Parameter: BOTtoken
13:04:26.386 -> *wm:[2] Added Parameter: timzon
13:04:26.386 -> *wm:[1] AutoConnect 
13:04:26.386 -> *wm:[2] ESP32 event handler enabled 
13:04:26.386 -> *wm:[2] Connecting as wifi client... 
13:04:26.386 -> *wm:[2] setSTAConfig static ip not set, skipping 
13:04:26.386 -> *wm:[1] Connecting to SAVED AP: Reis
13:04:26.912 -> E (2684365) wifi:sta is connecting, return error
13:04:26.912 -> *wm:[2] 60000 ms timeout, waiting for connect...
13:04:26.912 -> *wm:[2] . 
13:04:27.007 -> *wm:[2] . 

13:04:27.530 -> *wm:[2] [EVENT] WIFI_REASON:  201
13:04:27.530 -> *wm:[2] [EVENT] WIFI_REASON: NO_AP_FOUND 

13:05:26.915 -> *wm:[2] Connection result: WL_NO_SSID_AVAIL
13:05:26.915 -> *wm:[1] AutoConnect: FAILED 
13:05:26.915 -> *wm:[2] Starting Config Portal 
13:05:26.915 -> *wm:[2] Disabling STA 
13:05:26.915 -> *wm:[2] Enabling AP 
13:05:26.915 -> *wm:[1] StartAP with SSID:  deskpir
13:05:26.915 -> *wm:[2] AP has anonymous access! 
13:05:27.438 -> *wm:[1] AP IP address: 192.168.4.1
13:05:27.438 -> *wm:[1] Starting Web Portal 
13:05:27.438 -> *wm:[2] HTTP server started 
13:05:27.438 -> *wm:[2] Config Portal Running, blocking, waiting for clients... 
13:05:27.438 -> *wm:[2] NUM CLIENTS:  0
```

I think it would be useful to make the ESP restart or deepsleep for some minutes, then try again. Example: I am away from home during the week, there is no one there to login to the portal and reconnect; having the ESP deepsleep and try again later would at least give it a chance to go back online in case of router reboot.

Not sure how should I do it:

```
#define uS_TO_S_FACTOR 1000000ULL  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  300        /* Time ESP32 will go to sleep (in seconds) */

wm.setConnectTimeout(60); // how long to try to connect for before continuing
//wm.setConfigPortalTimeout(90); // auto close configportal after n seconds

#define uS_TO_S_FACTOR 1000000ULL  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  300        /* Time ESP32 will go to sleep (in seconds) */

  if (!wm.autoConnect(devname)) {
    Serial.println("failed to connect and hit timeout");
    delay(3000);

  // We set our ESP32 to wake up every 5 minutes
  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  Serial.flush(); 
  esp_deep_sleep_start();
  Serial.println("This will never be printed");
  }
```

Or maybe

```
#define uS_TO_S_FACTOR 1000000ULL  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  300        /* Time ESP32 will go to sleep (in seconds) */

  if (res) {
    Serial.println("Succesful Connection using WiFiManager");
    do_eprom_write();
  } else {

    //InternetFailed = true;
    Serial.println("Internet failed using WiFiManager - not starting Web services");
  // We set our ESP32 to wake up every 5 minutes
  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  Serial.flush(); 
  esp_deep_sleep_start();
  Serial.println("This will never be printed");
  }
```

jameszah, 2022-10-19:
https://github.com/jameszah/ESP32-CAM-Video-Telegram/blob/c5784535668f6836b2f187bad55afe016cfe78e3/v9.5/ESP32-CAM-Video-Telegram_9.5/ESP32-CAM-Video-Telegram_9.5.ino#L1297

You could just change this to 5-8 minutes.  
The cost is waiting 5 minutes when you move to a new wifi zone and you want to change the ssid. 
Could also add the button or power on-off-on sequence to send it back to ap mode... like in wifi lightbulbs.

FBMinis, 2022-10-19:
You are right, increasing to a few minutes is much more straightforward, thank you.

Nevertheless, out of curiosity and considering my examples above, where should I call the deep sleep? After the ConnectTimeout or InternetFailed = true;?

jameszah, 2022-10-21:
regarding the deepsleep stuff - after deepsleep the esp32 reboots, and you lose context ... unless you save it in nonvolatile memory ... so just delaying for a minute, then retrying, and repeat would be simpler.  You are not that focused on saving battery during that phase.

zilhos, 2022-10-22:
hello , can someone give me a hand with this please , i have erased the board before loading the software and try with a new board and the same error appears . Thank you

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
---------------------------------
ESP32-CAM Video-Telegram pir-cam 9.5.1
---------------------------------
PSRAM size to store the video 4194252
try to allocate 3682252
[E][camera.c:267] camera_fb_init(): Allocating 150 KB frame buffer Failed
[E][camera.c:1281] camera_init(): Failed to allocate frame buffer
E (1143) gpio: gpio_isr_handler_remove(398): GPIO isr service is not installed, call gpio_install_isr_service() first
[E][camera.c:1406] esp_camera_init(): Camera init failed with error 0x101
Camera init failed with error 0x101Camera Setup Failed!

jameszah, 2022-10-23:
https://github.com/jameszah/ESP32-CAM-Video-Telegram/blob/7b492ccc5af9c1f71c154e910b4cc6a63d544615/v9.5/ESP32-CAM-Video-Telegram_9.5/ESP32-CAM-Video-Telegram_9.5.ino#L1376

The camera init is failing because there is not enough psram (150kb framrebuffer).  The line above allocates as much as possible psram to record the video.  You need more psram for hd than for vga. (900 vs 500)  

You could bump that up to 1000 or 1200.  

I put a few versions in last night for hd and vga.  Are you compiling it, or using the precompiled version?  If you are compiling, you need to use the current compiler and arduino framework which will use different amounts of ram and psram.  Also it assumes the ai-thinker 4mb psram.

zilhos, 2022-10-23:
Thank you for your prompt response sir, that's right, the problem was there, put 1000 instead of 500 and it worked correctly. I think it's a cool project. thanks for sharing. Compile with the ide, although the automatic compilation works fine too