jameszah/ESP32-CAM-Video-Telegram #24: Esp32 receives only the first messages

Berta2001, 2022-09-17:
After connecting the device to the wifi, correctly writing the chat-id etc ... I sent the message / start from telegram and he received all the possible commands safely. After about half a day, esp32 no longer receives any commands as if it were blocked. To make it restart again I have to disconnect the power supply and reconnect it; so as to reboot it.
How can I solve this problem?
Thank you

jameszah, 2022-09-28:
How many messages is that - like dozens or hundreds or ????.  
Are there big bursts of messages?

Do you have the serial monitor during this problem?

FBMinis, 2022-09-28:
Poor signal; less than perfect connection to the power supply and/or multiple consecutive PIR false positives have cause my project to malfunction like that. Under adequate conditions, this program works reliably and is very versatile - I'm very thankful for James to have shared his work with us.

Reducing the sensitivity of the HC-SR501 PIR will help it not being triggered by the wind, plants rocking or door curtains dangling; 
Placing it in the shade or away from reflections should help too;
Increasing the ON time by positioning the Tx potentiometer at "9 o'clock";
Increasing the LOW time to 2minutes by changing C7 to 4,7uF (Ti value in BISS0001 datasheet);

If you dont need the motion detector feature, just dont plug it in. You can send recurring messages to the Telegram bot to retrieve a photo or a clip by using IFTTT and Macrodroid (free Android apps).

(I will edit this post later with some more info about my experiments)

Berta2001, 2022-10-03:
It usually occurs after a period of time of about one night; there are few messages in the order of tens.
What do you mean by serial monitor?

jameszah, 2022-10-03:
> 1. Create a macro in Macrodroid that tweets "{system time}" to your twitter account every 2min;
> 2. Create a Telegram group or channel comprised of yoursel, your camera bot and the IFTTT bot.
> 3. Create an applet in IFTTT that sends the word "/photo" to that group. You will receive a photo every 2min.

Nice - I'll give that a try

jameszah, 2022-10-03:
code to restore wifi after failing
https://github.com/jameszah/ESP32-CAM-Video-Telegram/blob/1b059b59dd29961eb4e3e1173db8489a2ce26b5e/ESP32-CAM-Video-Telegram_8.9x.ino#L1133

FBMinis, 2022-10-03:
> Nice - I'll give that a try

I've noticed an interval of 2min is too short for IFTTT, sometimes it wont check for new tweets for 10 min or more, then you receive 5 photos consecutively. It's been working fine for me with a 10min interval.

jameszah, 2022-10-04:
> I've noticed an interval of 2min is too short for IFTTT, sometimes it wont check for new tweets for 10 min or more, then you receive 5 photos consecutively. It's been working fine for me with a 10min interval.

Do you know if telegram has message limits?  
10 min or 6x24 jpgs per day is not too much, but at some point they might get annoyed ???

You could also add a line of code to send a image every 2 minutes to your id -- and still have enough free time for other commands.

I take it your scheme has ifttt triggered by the twitter message, and then it tells the esp32 to send the picture to telegram, so there is nothing (no image) on twitter in the end?

FBMinis, 2022-10-04:
> Do you know if telegram has message limits?
10 min or 6x24 jpgs per day is not too much, but at some point they might get annoyed ???

I've read there are limits in the order of a few gigabytes per day, with single file max size of 2Gb but unlimited cloud storage space. There's people running bots inside groups or channels with thousands of users, serving them movies, which is a much larger operation.

I have a channel that I use to test these ESP32-Cam projects, it's been online since March 2021 and it currently has almost 2500 photos. I've tried deleting them but Telegram client always downloads the thumbnails again, leading me to think they'll be in some server for a long time. Nowadays it's possible to have photos automatically deleted after 1 day (or 1 week, month or custom interval), but those 2500 precede this feature and seem undeletable. 

>You could also add a line of code to send a image every 2 minutes to your id -- and still have enough free time for other commands.

This would be an handy feature but your code is running so well I wont touch it because I'm just a hobbyist... 
Wifimanager with extra parameters would be a welcome feature, for example having a checkbox for PIR or Avi disabled on boot or how many minutes interval to automatically take the photo. I'm still reading about these libraries and experimenting with a Wemos D1.

> I take it your scheme has ifttt triggered by the twitter message, and then it tells the esp32 to send the picture to telegram, so there is nothing (no image) on twitter in the end?

Exactly. I setup Macrodroid to only tweet system time, as it's a variable that is constantly changing and twitter doesn't get suspicious this way. When IFTTT detects there's been a new tweet, it sends the instruction to the telegram channel through the IFTTT bot.

[image]