jameszah/ESP32-CAM-Video-Telegram #25: Got error, already used your custom UniversalTelegram setup

mikahoy045, 2022-09-24:
Hi
I received an error when compiling:
`/Users/mikah/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: sketch/Telegram_Bot.ino.cpp.o:(.literal._Z17handleNewMessagesi+0x154): undefined reference to `UniversalTelegramBot::sendMessage(String const&, String const&, String const&)'
Multiple libraries were found for "WiFi.h"
 Used: /Users/mikah/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/libraries/WiFi
 Not used: /private/var/folders/9d/r0y9dm4x1_9gqzgsl6d25l7w0000gn/T/AppTranslocation/0AE11984-4368-4370-A493-1D39A35D77EB/d/Arduino.app/Contents/Java/libraries/WiFi
/Users/mikah/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: sketch/Telegram_Bot.ino.cpp.o:(.literal._Z17handleNewMessagesi+0x15c): undefined reference to `UniversalTelegramBot::sendMultipartFormDataToTelegramWithCaption(String const&, String const&, String const&, String const&, String const&, String const&, int, bool (*)(), unsigned char (*)(), unsigned char* (*)(), int (*)())'
/Users/mikah/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: sketch/Telegram_Bot.ino.cpp.o: in function `handleNewMessages(int)':
/Users/mikah/Documents/Arduino/IsidorusCode/Telegram_Bot/Telegram_Bot.ino:312: undefined reference to `UniversalTelegramBot::sendMessage(String const&, String const&, String const&)'
/Users/mikah/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/mikah/Documents/Arduino/IsidorusCode/Telegram_Bot/Telegram_Bot.ino:330: undefined reference to `UniversalTelegramBot::sendMessage(String const&, String const&, String const&)'
/Users/mikah/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/mikah/Documents/Arduino/IsidorusCode/Telegram_Bot/Telegram_Bot.ino:407: undefined reference to `UniversalTelegramBot::sendMessage(String const&, String const&, String const&)'
/Users/mikah/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/mikah/Documents/Arduino/IsidorusCode/Telegram_Bot/Telegram_Bot.ino:421: undefined reference to `UniversalTelegramBot::sendMultipartFormDataToTelegramWithCaption(String const&, String const&, String const&, String const&, String const&, String const&, int, bool (*)(), unsigned char (*)(), unsigned char* (*)(), int (*)())'
/Users/mikah/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/mikah/Documents/Arduino/IsidorusCode/Telegram_Bot/Telegram_Bot.ino:453: undefined reference to `UniversalTelegramBot::sendMessage(String const&, String const&, String const&)'
/Users/mikah/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/mikah/Documents/Arduino/IsidorusCode/Telegram_Bot/Telegram_Bot.ino:502: undefined reference to `UniversalTelegramBot::sendMessage(String const&, String const&, String const&)'
/Users/mikah/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: sketch/Telegram_Bot.ino.cpp.o: in function `setup()':
/Users/mikah/Documents/Arduino/IsidorusCode/Telegram_Bot/Telegram_Bot.ino:1101: undefined reference to `UniversalTelegramBot::sendMessage(String const&, String const&, String const&)'
/Users/mikah/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: sketch/Telegram_Bot.ino.cpp.o: in function `send_the_picture()':
/Users/mikah/Documents/Arduino/IsidorusCode/Telegram_Bot/Telegram_Bot.ino:1168: undefined reference to `UniversalTelegramBot::sendMultipartFormDataToTelegramWithCaption(String const&, String const&, String const&, String const&, String const&, String const&, int, bool (*)(), unsigned char (*)(), unsigned char* (*)(), int (*)())'
/Users/mikah/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/mikah/Documents/Arduino/IsidorusCode/Telegram_Bot/Telegram_Bot.ino:1172: undefined reference to `UniversalTelegramBot::sendMultipartFormDataToTelegramWithCaption(String const&, String const&, String const&, String const&, String const&, String const&, int, bool (*)(), unsigned char (*)(), unsigned char* (*)(), int (*)())'
/Users/mikah/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: sketch/Telegram_Bot.ino.cpp.o: in function `send_the_video()':
/Users/mikah/Documents/Arduino/IsidorusCode/Telegram_Bot/Telegram_Bot.ino:1191: undefined reference to `UniversalTelegramBot::sendMultipartFormDataToTelegramWithCaption(String const&, String const&, String const&, String const&, String const&, String const&, int, bool (*)(), unsigned char (*)(), unsigned char* (*)(), int (*)())'
/Users/mikah/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: sketch/Telegram_Bot.ino.cpp.o: in function `loop()':
/Users/mikah/Documents/Arduino/IsidorusCode/Telegram_Bot/Telegram_Bot.ino:1118: undefined reference to `UniversalTelegramBot::sendMessage(String const&, String const&, String const&)'
collect2: error: ld returned 1 exit status`

Already copy and pasted it in the old UniversalTelegram bot and replace both file (*.cpp and *.h), any suggestion? Thanks in advance

FBMinis, 2022-09-24:
Delete Universal Telegram library; place the .h file and .cpp file inside the folder that contains ESP32-CAM-Video-Telegram_8.9x.ino

jameszah, 2022-09-28:
I think that is correct - I've changed my posting style to put things in a folder with the proper name that arduino insists upon.  Some changes coming to this one for various psram sizes.