jameszah/ESP32-CAM-Video-Telegram #26: Help

sinaj2889, 2022-10-10:
Yeah the picture shows the issue while compiling[image]

jameszah, 2022-10-11:
[image]

You need these 2 files in the same folder as the .ino file, and you have to include them in your project -- not sure which ide you are using there, but this is how it looks with arduino.

If you have the Brian Lough UniversalTelegramBot library installed, you might have to remove it, as I have added some functions that are not in the original.

FBMinis, 2022-10-11:
I think it is also important to have esp32 core v2.0.2 isntalled

sinaj2889, 2022-10-15:
OK ty

sinaj2889, 2022-10-15:
OK ty