jameszah/ESP32-CAM-Video-Telegram #30: esp32 not connect to bot

PloyPanmaneeCNC, 2022-11-10:
I upload program and setup via browser 
after turn on again bot in telegram is not ping status like your fast installer do

i want to add control lock command so i try upload by myself libraries i use that in 9.6 folder
[image]

jameszah, 2022-11-12:
https://github.com/jameszah/ESP32-CAM-Video-Telegram/blob/b9df4519c2aba9e8b74cc923e19858b832ec3b13/v9.6/ESP32-CAM-Video-Telegram_9.6/ESP32-CAM-Video-Telegram_9.6.ino#L1413

So it works with the one-click installer and fails with the changes, or that error is from the one-click-installer?

The 150KB line means the camera does not have enough memory, and the memory was used by the space used to record a video.  The line above reserves 500KB for the camera, but you could increase that to 700 ... 800.

Also it has only been tested with esp-arduino 2.04, arduino 1.8.19, and debug comments turned off.  These might impact the memory usage.

I cannot actually find the fb_init() line in the code just now.  Check your esp32-arduino version.

DJprasenjit, 2023-02-28:
use the board ai thinker esp32 board