jameszah/ESP32-CAM-Video-Telegram #31: old pic sent by the device

bmburi, 2022-11-13:
HI, every time the pir senses a motion the pic is sent is actually the last pic was taken before.
as well as it does the same when i ask for a new pic by command is there a way to fix this
thank u in advance!

bmburi, 2022-11-13:
@jameszah

jameszah, 2022-11-14:
what version are you using?
That happened with ver 8.8 (13 months ago), and corrected with 8.9 (9 months ago) after the esp32-arduino drivers changed.
The current 9.5 or 9.6 (1 month ago) should be okay ????

bmburi, 2022-11-14:
am using the latest version 9.6 
thank u so much for ur work btw 
its the latest version is there a way around it? @jameszah

jameszah, 2022-11-14:
Try this:
- at line 873, it takes a picture which will be sent as a still picture before the movie (which may be skipped if you are skipping movies)
- before that line, paste in the code below, which takes 4 pictures and discards them, deleting the inventory of old pictures that were added due to some driver changes

https://github.com/jameszah/ESP32-CAM-Video-Telegram/blob/b9df4519c2aba9e8b74cc923e19858b832ec3b13/v9.6/ESP32-CAM-Video-Telegram_9.6/ESP32-CAM-Video-Telegram_9.6.ino#L873

```
    for (int j = 0; j < 4; j++) {
      camera_fb_t * newfb = esp_camera_fb_get();
      if (!newfb) {
        Serial.println("Camera Capture Failed");
      } else {
        //Serial.print("Pic, len="); Serial.print(newfb->len);
        //Serial.printf(", new fb %X\n", (long)newfb->buf);
        esp_camera_fb_return(newfb);
        delay(10);
      }
    }
```

bmburi, 2022-11-14:
if its not too much trouble, is it possible to update the one link installation because i usually mess up the compiling!

bmburi, 2022-11-14:
i cant really compile it idk what am doing wrong

bmburi, 2022-11-15:
@jameszah  plus command clip does not work

FBMinis, 2022-11-16:
I don't have that old pic problem with this combination:

Arduino 1.8.13
Arduino core for ESP32 v2.0.2
ESP32-CAM-Video-Telegram v9.6

bmburi, 2022-11-16:
does the clip command works for u? 
do u have an sd card on?
i used tge one click installation did u u use the same?

FBMinis, 2022-11-17:
> i used tge one click installation did u u use the same?

I use the .ino file with the arduino IDE

> do u have an sd card on?

I do not use an sdcard

> does the clip command works for u?

Running v9.6, I get reboots and dont receive the video when in HD &&  /fast || /med setting. In HD && /slow || /vslow I receive the video with no reboots. Zero problem running v9.5 or v9.6 VGA.

I have twos cameras running for about 4 weeks now.