jameszah/ESP32-CAM-Video-Telegram #34: Video upload to telegram slow rate

mlegnari, 2022-11-16:
Hi! When ESP32CAM finishes the video and starts uploading I noticed that it takes a long time (150 seconds +- or 19kb/s) to upload the 2.9MB video.

I'm using very fast internet 300MB download/upload, at first I tried another connection without success, and I also tried 2 more modules from other suppliers.

Is this delay normal?

mlegnari, 2022-11-16:
@bmburi The pir works correctly, as soon as it detects it, it takes a photo of approximately 45kb in size, and sends it in a short time, 5 seconds approximately, soon after it films (45 seconds at 2fps) there it generates the 2.9MB file (fills all reserved PSRAM ) this upload is time consuming.

I also noticed that when there are packet losses (5% or more) the upload can be interrupted and does not finish, I tested it on a bad 4G internet and had this problem.

bmburi, 2022-11-16:
for me whenever senses motion the pic is old but the clip is okay

mlegnari, 2022-11-16:
@bmburi what would an old photo be? after the pir it shows you a photo on telegram approximately how many seconds ago?

For example: pir sensor triggered at 12:00h:00s, you will receive a photo on telegram at 12:00h:20s. Or are you getting a very old photo that has nothing to do with it?

bmburi, 2022-11-16:
lets say i send a command to take a picture.
it will come the pic.
after when it senses the pir it sends me that pic instead of the new one!

FBMinis, 2022-11-16:
That has happened to my while using AsyncTelegram2 library in another project. It was fixed by changing library and board version.

jameszah, 2022-11-30:
The "old-photo" issue is a old problem when the driver changed -- it is not fixed here apparently -- i think it is fixed the the test version.

The transfer speed bottleneck is between the esp32 and the router -- router to the telegram servers in Switzerland is probably quite fast.

I would guess the 3 MB file is 90 seconds or so.  The higher the rssi number the quicker it should go (higher = lower negative log number)

ESP32 is max like 3 mega-bits-per second, but doing the tsl processing and waiting for the reply from Switzerland slows things down.
  
Here is some chat about speed -- non-tls between esp32 and computer, through the router, in the same room.

https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior

https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior/raw/master/v58/2mbps.jpg

https://github.com/jameszah/ESP32-CAM-Video-Telegram/blob/b9df4519c2aba9e8b74cc923e19858b832ec3b13/v9.6/ESP32-CAM-Video-Telegram_9.6/UniversalTelegramBot.h#L142

Increasing the blocksize might help, but probably not enough free ran to do much.

bmburi, 2022-12-01:
its fixed for me by the code u made change
the old pic issue

FBMinis, 2023-02-14:
Regarding the slow upload, I recently purchased a 6th camera that had trouble connecting to wifi and send picture+video. I could see the led blinking while capturing but the Serial Monitor showed the camera upload would end and I'd receive nothing. The test conditions were the same as with the previous 5. I decided to keep it simple and test with the CameraWebServer example in the ArduinoIDE. What I observed was a difficulty to connect, extremely low framerate and sudden crashes.

Both the CameraWebServer example and JamesZah worked normally after I adopted this workaround:

https://github.com/espressif/arduino-esp32/issues/4655
https://github.com/espressif/arduino-esp32/issues/5834
https://community.m5stack.com/topic/3649/unitcam-hw-issue-cannot-stream-video/11
https://www.youtube.com/watch?v=MBOGWK5qrkw

It affects video playback but  I got a refund from the seller: https://pt.aliexpress.com/item/1005004749706289.html (purchased the version without lens for €4.30 at the time)

hakam007, 2023-08-19:
> its fixed for me by the code u made change the old pic issue

please help in this regard ..I am still facing the old pic issue

jameszah, 2023-08-19:
Are you using the installer, or compiling the 9.6 version?
Which esp32-arduino board library?
I don't seem to have updated since 2.06 ... or 2.09 on another ide .... now it is 2.11

I'll have a look at it.

hakam007, 2023-08-19:
> Are you using the installer, or compiling the 9.6 version?
> Which esp32-arduino board library?
> I don't seem to have updated since 2.06 ... or 2.09 on another ide ....
> now it is 2.11
>
> I'll have a look at it.
thanks a lot sir, I modified the code that you suggested in GitHub , now it
works fine , you are really doing great work 👏👏🙏🙏😍😍