jameszah/ESP32-CAM-Video-Telegram #35: PIR detected

bmburi, 2022-11-28:
is there a way while the **AVI is enabled** to send a message saying that **motion detected** and send u the video instead of what is happening now which sends the pic that says pir event and u get the video 
thanks in advance!

FBMinis, 2022-11-28:
> is there a way, while the AVI is enabled, to send a message saying that motion detected and send u the video, instead of what is happening now, which sends the pic that says pir event and u get the video

Meaning, you don't want to receive the photo before the video? You only want to receive a message before the video?
 
I would comment line 1494 `send_the_picture();` and paste under it `bot.sendMessage(chat_id, "motion detected", "");`

bmburi, 2022-11-28:
well now it sends me the picture the message and the video 
how can i make it only the message and the video?

bmburi, 2022-11-28:
plus on enavle pir it sends only the message not the pic?

jameszah, 2022-11-30:
Out of town for a while ... catching up on discussions

The pir interrupt starts a task, which sets a variable:

https://github.com/jameszah/ESP32-CAM-Video-Telegram/blob/b9df4519c2aba9e8b74cc923e19858b832ec3b13/v9.6/ESP32-CAM-Video-Telegram_9.6/ESP32-CAM-Video-Telegram_9.6.ino#L879

Then in the loop, it checks the variable, and does something

https://github.com/jameszah/ESP32-CAM-Video-Telegram/blob/b9df4519c2aba9e8b74cc923e19858b832ec3b13/v9.6/ESP32-CAM-Video-Telegram_9.6/ESP32-CAM-Video-Telegram_9.6.ino#L1491 

In the code it calls send-the-picture, instead, put in this code

https://github.com/jameszah/ESP32-CAM-Video-Telegram/blob/b9df4519c2aba9e8b74cc923e19858b832ec3b13/v9.6/ESP32-CAM-Video-Telegram_9.6/ESP32-CAM-Video-Telegram_9.6.ino#L375

```
    String hi = "Got a pir interrupt ";
    bot.sendMessage(chat_id, hi, "Markdown");
    client.setHandshakeTimeout(120000);
```

You also need to delete the picture that would be sent -- this line from send-the-picture

`esp_camera_fb_return(vid_fb);`

... that was allocated in the camera_loop

`vid_fb = get_good_jpeg(); `

hakam007, 2023-08-19:
hello sir please guide me and I will be highly thankful to you....
whenever any PIR event happened thn its send the pic, but the pic is not live it is long time ago taken. please help in this regard