jameszah/ESP32-CAM-Video-Telegram #42: Move the ESP to deepsleep but waiting for PIR event?

sergiocasero, 2023-01-31:
Hello, first of all, congrats, the sketch works so so good.

My question is: Is there any way to "disable" everyting except the PIR and the video transfer? I want to put the ESP in a place where is no power supply so I only have a 18650 battery, the idea is to have the ESP in deepsleep until PIR event is fired -> send video -> go deep sleep again.

What do you think?

FBMinis, 2023-01-31:
The ESP-32CAM has a high deepsleep current, I found it's better to keep it OFF while there is no movement.

I set the PIR to stay ON for 3min. When it detects movement, the output goes HIGH, which powers the Camera throught the P-Mosfet. It sends me a photo with the RSSI value as caption; then sends the video; then sends a message with commands to take another photo, capture another clip or turn the flash ON (this might be redundant, so it's optional - I'm still testing). After 3min the power is turned OFF.

I keep this camera outside, where I dont have a 220V line but it's still under wifi range.

The project consumes 272uA of current while nothing is happening because the PIR is ON and I have a 0.9V to 5V boost module (power from 3xAAA)

[image]
[image]

https://github.com/jameszah/ESP32-CAM-Video-Telegram/discussions/33#discussion-4570513

sergiocasero, 2023-02-01:
Thanks you so much, @FBMinis what is the part number of the mosfet you're using?

FBMinis, 2023-02-01:
The mosfets I'm using are the AO3400 and AO3401.

sergiocasero, 2023-02-02:
> The mosfets I'm using are the AO3400 and AO3401.

I'm trying to find them in local stores but without success, any alternatives?

FBMinis, 2023-02-02:
This is what I have in my notes:

[image]

You can search for AO3400 in websites like Mauser, Digikey, LCSC and then select other mosfet with similar characteristics. Maybe also google "logic level Mosfet" or look at the schematic of open source hardware like the ESP-32CAM.