jameszah/ESP32-CAM-Video-Telegram #49: wifi-manager issue, and photo quality

JackFrap, 2023-08-16:
Hello,
1- I set the wifi parameters through the wifi manager. But sometimes when I turn on the device it goes back to wifi-manager mode by itself and I have to re-enter the wifi parameters again. Is it possible to disable the wifi manager and enter the wifi parameters directly from the code? Thank you
2- How can I set the code so that with the "/photo" command it sends the photo with the highest possible image quality?

Thank you very very much!

JackFrap, 2023-08-22:
1- I think maybe I solved the first problem by eliminating 4 lines in the code: 1486-1487-1488-1489
… if (reset_request) {  //  wm.resetSettings();  //  reset_request = false; //  } …

2- Can you tell me which lines of code I have to modify to send photos to telegram in the highest possible quality, I'm not interested in videos, just photos Thank you 🙏

jameszah, 2023-08-22:
https://github.com/jameszah/ESP32-CAM-Video-Telegram/blob/b9df4519c2aba9e8b74cc923e19858b832ec3b13/v9.6/ESP32-CAM-Video-Telegram_9.6/ESP32-CAM-Video-Telegram_9.6.ino#L1337

The connection issue is in the wifimanager - I think it tries to connect to your wifi, but fails and then it runs the system to give you another opportunity to set your wifi.

A few lines of code to retry your wifi connection would help.

JackFrap, 2023-08-22:
Thank you!
After some testing these are the changes i made to set the quality to uxga:

159 framesize_t configframesize = FRAMESIZE_UXGA; // FRAMESIZE_ + QVGA|CIF|VGA|SVGA|XGA|SXGA|UXGA
160 int frame_interval = 0;          // 0 = record at full speed, 100 = 100 ms delay between frames
161 float speed_up_factor = 1;          // 1 = play at realtime, 0.5 = slow motion, 10 = speedup 10x
162 int framesize = FRAMESIZE_UXGA; //FRAMESIZE_HD;
163 int quality = 5;
164 int qualityconfig = 5;

1415  if (hdcam) {
1416    avi_buf_size = avail_psram - 1600 * 1200; //900 for hd, 500 for vga
1417    configframesize = FRAMESIZE_UXGA; // FRAMESIZE_ + QVGA|CIF|VGA|SVGA|XGA|SXGA|UXGA
1418    framesize = FRAMESIZE_UXGA; //FRAMESIZE_HD;
1419    Serial.println("Camera to UXGA ");

closed 2023-08-22 by JackFrap