jameszah/ESP32-CAM-Video-Telegram #52: Send Video to multiple chat id?

alfa934, 2023-09-03:
I've edited the code so that once it records the video, it will send to multiple chat id. So far, I've only been successful at sending messages. Every time I try to send the video to multiple id, it only gets sent to one chat. I'm a novice programmer so I don't quite understand the whole code, but is it possible to send a video to multiple chat id? Thank you 🙏

alfa934, 2023-09-03:
Nevermind, I was able to create a new function and send it to 2 users. Even though it's a simple solution

closed 2023-09-03 by alfa934

jameszah, 2023-09-04:
So do you just run the send twice with different chat_id?
You can set up a chat group I think with multiple id's.
I dont think you can specify multiply recipients in a send ????

alfa934, 2023-09-06:
> So do you just run the send twice with different chat_id? You can set up a chat group I think with multiple id's. I dont think you can specify multiply recipients in a send ????

Yeah, that's basically what I did. I made a copy of the send video function but with a for loop repeating the whole thing and it iterates over a list of id chat. So, i have chat_id (main id) and chat_id_list to send to other id. But because of that, it's very slow in sending, but it works xD 

I tried using a group chat but it didnt send the video to the group, but maybe I didnt write it properly. Worth another try