jameszah/ESP32-CAM-Video-Telegram #54: eror

kieerz7, 2024-01-11:
[image]

jameszah, 2024-01-11:
That is the web installer file.
You go the the website and click "Connect" and it installs that file and a few others onto your esp32.
Lengthy instructions here (for a different program)
https://jameszah.github.io/ESP32-CAM-VideoCam/instruct

kieerz7, 2024-01-11:
i need to do the same with esp32-cam-video telegram-main if not can u give me instruction on it

kieerz7, 2024-01-12:
> That is the web installer file. You go the the website and click "Connect" and it installs that file and a few others onto your esp32. Lengthy instructions here (for a different program) https://jameszah.github.io/ESP32-CAM-VideoCam/instruct

can i use hotspot instead of wifi

kieerz7, 2024-01-12:
i already done intalling but im stuck at "After installing:
connect your phone or computer to wifi device “deskpir”" because i im using hotspot