jameszah/ESP32-CAM-Video-Telegram #55: strange things in code 9.6

jmn4472, 2024-02-08:
Hello,
In the 9.6 version 
I read line 179 and 180
**String chat_id** = "1234567890";
**#define BOTtoken** "XXXXXXXXX:XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX"  // your Bot Token (Get from Botfather)
then line 190 and 193
**String chat_id;**
//#define BOTtoken "XXXXXXXXX:XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX"  // your Bot Token (Get from Botfather)
**String BOTtoken;**

So, what is the right one?
If I use #define BOTtoken, that generate error a bit further in the compile.
Regards, JM

FBMinis, 2024-02-09:
Did you modify the code?

From V9.5 onwards, the values for chat_id and BOTtoken are entered in Wifimanager, therefore like 171 to 181 are commented out.

```
// Initialize Wifi connection to the router and Telegram BOT
/*
  char ssid[] = "yourssid";     // your network SSID (name)
  char password[] = "yourssidpassword"; // your network key
  // https://sites.google.com/a/usapiens.com/opnode/time-zones  -- find your timezone here
  String timezone = "GMT0BST,M3.5.0/01,M10.5.0/02";

  // you can enter your home chat_id, so the device can send you a reboot message, otherwise it responds to the chat_id talking to telegram

  String chat_id = "1234567890";
  #define BOTtoken "XXXXXXXXX:XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX"  // your Bot Token (Get from Botfather)
*/
char ssid[] = "ssid1234";     // your network SSID (name)
char password[] = "mrpeanut"; // your network key
// https://sites.google.com/a/usapiens.com/opnode/time-zones  -- find your timezone here
String timezone = "MST7MDT,M3.2.0/2:00:00,M11.1.0/2:00:00" ; //"GMT0BST,M3.5.0/01,M10.5.0/02";

// you can enter your home chat_id, so the device can send you a reboot message, otherwise it responds to the chat_id talking to telegram

//String chat_id = "1234567890";
String chat_id;
//#define BOTtoken "XXXXXXXXX:XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX"  // your Bot Token (Get from Botfather)
String BOTtoken;
```

jmn4472, 2024-02-09:
No, I didn't modify the code, your code. 
I had not understood that the chat_id and the BOTtoken are integrated by wifimanager. 
Sorry for the time you loose.

That's a very good projet, you did.
Regards

closed 2024-03-01 by jmn4472