jameszah/ESP32-CAM-Video-Telegram #56: /Flash

MamoNieTeraz, 2024-02-28:
Hi,
Your project is great, and it is very helpful for me to improve the already created product which is the automatic feeder. The only thing I'm missing is some functionality, namely lighting up the LED when motion is detected to keep the recording lit. Is it possible to introduce this functionality in the future ? 
Warm regards !!

FBMinis, 2024-03-04:
I've solved the problem of capturing video in dark settings when there is movement with one of these: aliexpress.com/item/1005005727625716.html
aliexpress.com/item/1005006399943906.html

They're well built and the light is strong. I bought one recently for under €7, you just need to browse around because Aliexpress prices are all over the place nowadays.

FBMinis, 2024-03-04:
```
#define LAMP_PIN 4

int lampChannel = 7;          // a free PWM channel (some channels used by camera)
const int pwmfreq = 50000;    // 50K pwm frequency
const int pwmresolution = 9;  // duty cycle bit range
const int pwmMax = pow(2, pwmresolution) - 1;

// Lamp Control
void setLamp(int newVal) {
	if (newVal != -1) {
		// Apply a logarithmic function to the scale.
		int brightness = round((pow(2, (1 + (newVal * 0.02))) - 2) / 6 * pwmMax);
		ledcWrite(lampChannel, brightness);
		Serial.print("Lamp: ");
		Serial.print(newVal);
		Serial.print("%, pwm = ");
		Serial.println(brightness);
	}
}

void setup() {

	pinMode(LAMP_PIN, OUTPUT);                       // set the lamp pin as output
	ledcSetup(lampChannel, pwmfreq, pwmresolution);  // configure LED PWM channel
	setLamp(0);                                      // set default value
	ledcAttachPin(LAMP_PIN, lampChannel);            // attach the GPIO pin to the channel

}

// As as example, when taking a Timed Photo
void loop() {

	if (millis() > (TimePhoto_lasttime + TimePhoto_Minutes * 60000) ) {
		if (tim_enabled) {
			for (int j = 0; j < 4; j++) {
				camera_fb_t * newfb = esp_camera_fb_get();
				if (!newfb) {
					Serial.println("Camera Capture Failed");
				} else {
					//Serial.print("Pic, len="); Serial.print(newfb->len);
					//Serial.printf(", new fb %X\n", (long)newfb->buf);
					esp_camera_fb_return(newfb);
					delay(10);
				}
			}

			setLamp(100);

			camera_fb_t * tfb = esp_camera_fb_get();

			setLamp(0);

			if (!tfb) {
				Serial.println("Camera Capture Failed");
			} else {

				currentByte = 0;
				fb_length = tfb->len;
				fb_buffer = tfb->buf;

				Serial.println("\n>>>>> Sending timed photo, bytes=  " + String(fb_length));

				String sent = bot.sendMultipartFormDataToTelegramWithCaption("sendPhoto", "photo", "img.jpg",
				              "image/jpeg", "Timed Photo", chat_id, fb_length,
				              isMoreDataAvailable, getNextByte, nullptr, nullptr);

				Serial.println("done!");
			}
			esp_camera_fb_return(tfb);
			delay(10);
		}
		TimePhoto_lasttime = millis();
	}

}
```