jameszah/ESP32-CAM-Video-Telegram #57: 1st second in AVI is from a previous video

jmn4472, 2024-04-01:
Hi all,
All is working correctly with that fabulous project, exception with the AVI files sent to Telegram. It seems that the 1st second is coming from a previous AVI (a bit hard to certify). I see clearly a change in what is recorded after one second (or around one second). For example a person is present in the 1st second but suddenly has disappeared in the rest of the video. 
I used INO 9.6 IDE 1.8.19 with AI ThinkerESP32 CAM board selected. Partition scheme is "Huge APP 3MB NO OTA/1MB SPIFFS". 
In the Cam, the speed to record is the default. 
The board is not from AI Thinker but from Aliexpress. Could the issue be due to the use of the wrong board selected? 
Regards

jameszah, 2024-04-02:
I take it that you compiled and downloaded it.  There is a issue with the defaults changing in the newer versions of the library.
You could add these six lines to fetch and discard the old images in the queue.  Just before this line about "should take zero time".

https://github.com/jameszah/ESP32-CAM-Video-Telegram/blob/b9df4519c2aba9e8b74cc923e19858b832ec3b13/v9.6/ESP32-CAM-Video-Telegram_9.6/ESP32-CAM-Video-Telegram_9.6.ino#L889

```
   fb_next = get_good_jpeg();
   esp_camera_fb_return(fb_next);
   fb_next = get_good_jpeg();
   esp_camera_fb_return(fb_next);
   fb_next = get_good_jpeg();
   esp_camera_fb_return(fb_next);

    fb_next = get_good_jpeg();               // should take zero time    
    last_frame_time = millis();
    start_avi();
```

jmn4472, 2024-04-02:
Yes, compiled and working very well but this behaviour. I will install these lines next days.
On another project named "ESP32-CAM-Video-Recorder" you recommand to install C1 220µF between +5V and GND on the ESP32 and a 10K resistor IO13 and Vout of the PIR. Do you think these are still mandatory?

jameszah, 2024-04-02:
The capacitor is to prevent brownouts - if you have a good power supply, you can skip it.
The 10k is to reduce the current from the pir, as we just need the voltage level, not the current.
Some of those pir boards will give you lots of current to drive lights or big circuits.
I think it is more necessary if you are using an sd card, as the sd card is tied to io13.