jameszah/ESP32-CAM-Video-Telegram #58: Error when verifying Compile

jimthedj65, 2024-05-31:
I have revisited this after a few years and for some reason on Arduino 2.3.2 I get the following error when **verify/compile**

Using ESP-EYE with correct pinouts and correct led pin 22 for the flash.

line 438, 1185, 1189, 1208 `Compilation error: 'class UniversalTelegramBot' has no member named 'sendMultipartFormDataToTelegramWithCaption'; did you mean 'sendMultipartFormDataToTelegram'?` 

Any idea what this could be, is it related to a newer version of Arduino

jameszah, 2024-06-02:
I think you have the UniveralTelegramBot library installed -- which does not have the ..withCaption routine, instead of bringing all the modules of the program -- which does have the ..withCaption routine.

[image]

Have you also replaced the quotes with the angle brackets?

```
//#include <UniversalTelegramBot.h>
#include "UniversalTelegramBot.h"  // use local library which is a modified copy of an old version
```

There is a much newer version with a pre-compiled installer, but no dicumentation for that yet.  I suppose it wouldn;t work with the esp-eye anyway.

jameszah, 2024-06-02:
Same issue with ver 8.9 -- dont think arduino version is an issue, other than it has its own libraries.