jameszah/ESP32-CAM-Video-Telegram #61: Have reboots after every radu callback v.11.16

CHE77, 2024-10-12:
Have 2 cams with SD card. All work.[image]

But have reboots after every radu callback v.11.16

jameszah, 2024-10-12:
Is there anything on serial monitor?
It seems I did not post the code for that -- I'll have to track it down.

jameszah, 2024-10-12:
very low radu numbers extend the exposure time to about 4 seconds ... if it goes over 4 seconds there are problems with the espressif camera drivers.

You could try not lowering it below 8 maybe. /radu8

CHE77, 2024-10-13:
Noted that reboots are not related with low radu, but with it's recalculation.
[image]

Also, when I set radu value manually esp starts reboot continuously 
[image]

jameszah, 2024-10-13:
That final picture is just the same message arriving over and over at the esp from telegram - it is not fully received and acknowledged so it comes again after the reboot - you have to clean out the queue of messages.